aquariumbio/aquarium #321: AQ dynamic reassignment of items sometimes undesirable

dvnstrcklnd, 2018-09-21:
As a researcher, it is often necessary to submit plans with some operations using a specific Item. For example, when planning NGS preps, it is critical that the plan use all of the output items from a previous plan, and that each one be predictably assigned to a chain of operations so that they can be paired with the correct barcodes. However, when an input is clicked in the graphical planner, the software automatically resets the item to the first available. This may also happen at other places in the software.

It would be nice if field values could be flagged so that the software does not attempt to automatically assign the item. To be workable, individual instances of fvs would probably need to be addressable. Setting the behavior in the Operation Type definition probably wouldn't be satisfactory.

dvnstrcklnd, 2018-09-22:
It appears that this can also happen after the relveant operations are completed. For example:

From the Aquarium Job record:
```
Job 80383 performed by aidopotato on Sep 17, 2018

8:30 am - Treat samples with zymolyase
Add zymolyase to each sample according to the table
Yeast Library Item ID (input)	zymolyase (µl)
157152	5
157150	5
157148	5
157146	5
157144	5
156819	5
157704	5
158043	5
158041	5
158039	5
157710	5
157708	5
157706	5
158045	5
```

Looking at the relevant field values using trident:
```
plan = nav.session.Plan.find(25231)
fvs = [op.input("Yeast Library").updated_at for op in plan.operations if op.operation_type.name == "Treat With Zymolyase"]
print(fvs)

['2018-09-19T08:31:27.000-07:00', '2018-09-19T08:31:28.000-07:00', '2018-09-19T08:31:29.000-07:00', '2018-09-19T08:31:29.000-07:00', '2018-09-19T08:31:30.000-07:00', '2018-09-19T08:31:31.000-07:00', '2018-09-19T08:31:31.000-07:00', '2018-09-19T08:31:32.000-07:00', '2018-09-19T08:31:33.000-07:00', '2018-09-19T08:31:34.000-07:00', '2018-09-19T08:31:34.000-07:00', '2018-09-19T08:31:35.000-07:00', '2018-09-19T08:31:36.000-07:00', '2018-09-19T08:31:36.000-07:00', '2018-09-19T08:31:37.000-07:00', '2018-09-19T08:31:37.000-07:00']

plan = nav.session.Plan.find(25231)
fvs = [op.input("Yeast Library").item.id for op in plan.operations if op.operation_type.name == "Treat With Zymolyase"]
print(fvs)

[156820, 156820, 156820, 156820, 156820, 156820, 156820, 156820, 156820, 156820, 156820, 156820, 156820, 156820, 158042, 158040]
```
I believe that this probably happened as Sam was replanning some downstream operations and was looking back at the original plan for reference. 

NB, this reassignment broke the provenance for this job.

klavins, 2018-09-27:
I am trying to replicate this with a simple example and can't. When I make a plan in the Designer and choose something other than the first item for an input, I am able to save the plan, launch it, run it from manager, and then reload it in designer and my choice of item is always maintained. 

Can you come up with a sequence steps that leads to incorrect results? The problem with describe something that already happened, is that I don't know the sequence of steps that got you to the point where the item associations are wrong or where it first went wrong.

dvnstrcklnd, 2018-09-27:
Is the choice maintained after you click on the input?