aquariumbio/aquarium #322: Routing Behavior in Planner

jonluntzel, 2018-09-24:
When wiring an input to an output with multiple possible Sample Type / Object Type combinations, and one of those matches with the input's ST / OT, an orange wire indicating a mismatch between them still appears. 

For example, although Rehydrate Fragment produces a ST/OT of Fragment / Fragment Stock, and Make PCR Fragment accepts this combination as an input, when you try to wire them together, an orange wire appears - and - an additional bug is that the wire does not point from the initial operation’s output to the next operation’s input, but instead wires from the next operation’s input to the input of the previous operation.

Ideally, if out of all Sample and Object Type possibilities there exists an acceptable combination, it should default both ST / OT to those values. Prioritizing those combinations with a field value which has already been satisfied would make sense. A change on one ST / OT choice should propagate to any linked FV connected by wire or routing ID.

jonluntzel, 2018-09-24:
[image]

klavins, 2018-12-21:
Fixed in 2.5.0

closed 2018-12-21 by klavins