aquariumbio/aquarium #348: Error: Creating a new collection

frba, 2019-04-16:
I created a new container
--
Plate used for PCR, Moclo and OligosMin/Max: 0/100Unit: ul($0.01 each)Handler: 16×24 collectionRelease(Return):Safety: No safety informationCleanup: No cleanup informationData:(Could not parse as JSON)Vendor: No vendor informationLocation Prefix: Bench
--

However, when I try to create a new collection at samples menu I got an error at the browser. Am I doing something wrong?

--
VM2536:1 PUT http://0.0.0.0:8080/collections/1 500 (Internal Server Error)
(anonymous) @ VM2536:1
(anonymous) @ angular.js:13531
s @ angular.js:13257
(anonymous) @ angular.js:12998
(anonymous) @ angular.js:17914
$digest @ angular.js:19075
$apply @ angular.js:19463
(anonymous) @ angular.js:28819
dispatch @ jquery.min.js:2
y.handle @ jquery.min.js:2

angular.js:15536 Possibly unhandled rejection: {"data":"<!DOCTYPE html>\n\n<html>\n  <head>\n\n    <title>Internal Error (500)</title>\n\n    <style type=\"text/css\">\n\n      body { \n        background-color: #fff;\n        color: #666;\n        text-align: center;\n        font-family: arial, sans-serif;\n        padding: 0;\n        margin: 0;\n      }\n      .warn {\n        width: 100%;\n        margin-top: 100px;\n        text-align: center; \n        padding: 0;\n      }\n      h1 {\n        font-size: 28px;\n        color: #aaa;\n        line-height: 1.5em;\n      }\n      .logo {\n        margin: 0;  \n        width: 100%;\n        font-size: 32px; \n        text-decoration: none;\n        text-transform: uppercase;\n        color: rgb(0,255,34); \n        font-weight: bold;    \n        background-image: linear-gradient(rgb(24, 86, 148), rgb(16, 108, 200));\n        text-align: left;  \n        padding: 6px;           \n        a {\n          color: darkblue;\n          text-decoration: none;  \n          &:hover {\n            text-decoration: none;\n          };                    \n        }\n      }\n\n    </style>\n\n  </head>\n\n  <body>\n\n    <div class=\"logo\">\n      <a href=\"/\" class=\"logo\">Aquarium</a>\n    </div>\n\n    <div class=\"warn\">\n      [image]\n      <br><br>\n      <h1>Aquarium encountered a problem.</h1>\n      <p>If you can reproduce the conditions that led to this error, the developers would \n         appreciate a <a href=\"https://github.com/klavinslab/aquarium/issues\">bug report</a>.\n    </div>\n\n  </body>\n\n</html>\n\n","status":500,"config":{"method":"PUT","transformRequest":[null],"transformResponse":[null],"jsonpCallbackParam":"callback","url":"/collections/1","headers":{"Accept":"application/json, text/plain, */*","X-XSRF-TOKEN":"1BzjO8jUyQdENCUHRHSi4q2e9mRLl0HE7e5xWw4G4CM4K4B9udlLGWP/TNPXt4NZMSfTXmH761FZ5Y71GCQmhA=="}},"statusText":"Internal Server Error","xhrStatus":"complete"}

bjkeller, 2019-04-16:
`Unit` actually should refer to a single item, but this is not the problem.

I made a similar example and did not get the error.

Can you give me more detail on what happens when you create the collection?  E.g., what you see when you click the `new collection` button, what you choose, and when exactly the error occurs.

bjkeller, 2019-04-16:
Also, in what environment are you running Aquarium?

frba, 2019-04-16:
After clicking in the new collection it shows the name of the container: 384 plate. Then I click on this option and I got the error.

frba, 2019-04-16:
If the wizard is empty. Can we have an error? 
I just noticed once difference between plates with and without Bench. It makes sense?

bjkeller, 2019-04-16:
Not unless you are able to create a collection for a different type. Is that the case?

bjkeller, 2019-04-16:
Please run the following command `docker ps --no-trunc` in a separate terminal, and share the output.

Also, is this using the same database that you were populating with mysql commands?

frba, 2019-04-17:
> Not unless you are able to create a collection for a different type. Is that the case?
> 
> But, actually, no, this shouldn't result in an error.

What do you mean about a different type?

frba, 2019-04-17:
> Please run the following command `docker ps --no-trunc` in a separate ubuntu terminal (not within docker), and share the output.
> 
> Also, is this using the same database that you were populating with mysql commands?

I didn't try to populate the database using mysql, only uploading .csv files. I used mysql only to consult.

docker ps --no-trunc
[image]

bjkeller, 2019-04-17:
Please send me the database dump to my email so that I can see if it replicates on a local image.

frba, 2019-04-17:
I created a new project using the master branch. 
At first, the database has some containers on it (part and sample_container), I keep them there and I created a new container called 96 plate (collection), create a new sample type and I was able to create a new collection and see the plate table. 

At second I did: sudo rm -rf docker/db/*
Open the aquarium, I deleted the containers (part and sample_container) that starts with the db, create the 96 well plate and sample, then the new collection didn't work anymore.

At third I did: sudo rm -rf docker/db/*
Open aquarium, create the 96 well plate and sample, open the collection create the plate. After that I remove the containers (part and sample_container) and I tried again to create a new collection and it works.

bjkeller, 2019-04-17:
This is a completely different issue. It *should* be that if you delete the `docker/db/*` files that the database should be completely reinitialized, but I have also occasionally seen this sort of behavior.  It appears to be due to the way that I used shared volumes in the compose file, but has nothing to do with Aquarium itself. 

Also, those part and sample_container definitions should be considered built-in and you shouldn't delete them.

bjkeller, 2019-04-17:
OK, I was able to get the failure to occur with your dump.  The problem seems to be that you deleted the built-in definitions and when you try to create a collection, Aquarium is looking for the definition of `__Part` and not finding it.

I believe that your definition of the `part` sample type and the built-in usage of `part` as a part of a collection could coexist.  Though obviously there would be opportunities for confusion.  So, I wonder if you could use a more descriptive name than `part`, which has a lot of possible meanings. Perhaps, something like `start_site`.

I'm afraid the most reliable approach to get past this would be to start over with a fresh database.

bjkeller, 2019-04-17:
I'm going to go ahead and close this.  If you have questions about how to proceed, just send an email.

closed 2019-04-17 by bjkeller