aquariumbio/aquarium #364: Designer switch dialog "switch" button should use standard highlight color

bjkeller, 2019-07-18:
In the designer tab, when you switch between plans a dialog opens. The "switch" button should be the dark blue highlight color, but is grey.

lekhacminhphuong, 2019-07-18:
``` js
var dialog = $mdDialog.confirm()
              .title('Switch Plan?')
              .textContent(msg ? msg : "The current plan will be saved")
              .ariaLabel('Save First?')
              .ok('Switch')
              .cancel('Cancel')
              .clickOutsideToClose(true); 

 $(".md-primary.md-confirm-button.md-button.md-ink-ripple.md-default-theme").attr('id', 'switch_dialog')
```
I use Jquery to access to the "Switch" button, adding "switch_dialog" id attribute for the button, and using this id to style the button in browser.scss.

However, I did not get the color change on the "Switch" button.

lekhacminhphuong, 2019-07-22:
Fixed by d8658aa

lekhacminhphuong, 2019-07-22:
Fixed by 4898ab8

closed 2019-07-22 by lekhacminhphuong