aquariumbio/aquarium #379: Lab Specific Instrument Library

malloc3, 2020-02-05:
Within a protocol I need to know what instruments a lab has in order to use the correct instrument libraries.  Currently this is hard coded and the protocol has to be changed in order to work with different labs.  Reference the Hight Throughput Culturing protocols (specifically HighThroughputHelper, HighThroughputConstants, and HighThroughputSoftware).

dvnstrcklnd, 2020-08-26:
@malloc3 can you elaborate on this? Has your thinking changed since working on the COVID protocols?

malloc3, 2020-08-26:
My thinking has changed some yes.    The HTC protocols have complicated machine requirements etc and the way its implemented works well enough for that application.

The way that the Covid protocols use PCR is a lot better (for now this may be my "go to" method).   I really like the use of the "Parameters" section for a lab to state what equipment they have.     Although each piece of equipment may need a specific library added to the code I could foresee a time when we have a large machine library base.   In that case they would just need to install the right library and state in parameters what machines they have (a lot easier and less intimidating than changing the JSONs in the HTC protocols).