aquariumbio/aquarium #390: Syntax in op.temporary[] dictionaries is different in test than in real manager scenario[BUG]

XavaarQ, 2020-03-04:
**Describe the bug**
##If you have this
key = "key"
table operations.start_table
.get(key, type: 'number', heading: "example", default: -1) 
.end_table

##This will happen while testing
print(op.temporary[:key].to_f) => -1.0
print(op.temporary["key"].to_f) => 0.0

#But when run the exact same protocol in the real manager scenario this will happen
print(op.temporary[:key].to_f) => 0.0
print(op.temporary["key"].to_f) => -1.0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Nursery Server => Developer => A-Duckweed => Duckweed Culture Split 
2. Test the Protocol 
3. Notice the 3rd to last Show block, lines 163 to 173 in the protocol
4. Run this same protocol manually through the Design and Manager tabs, enter what ever number you want when prompted, Notice the reversal of what happened in the test scenario

**Expected behavior**
I expect this to function the same in both the Text and production versions

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]
[image]

**Computer/Device (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

bjkeller, 2020-03-06:
Thanks for the report. I changed the handling of table inputs in the test execution so that it works the same way as normal execution.  This won't show up in production until the changes are pushed.

A couple of things to note:
1. The default behavior during normal operation is to convert the key to a symbol. 
    So, `:key` instead of`"key"`.

    I'll admit it is a bit confusing to give the key as a string and have to use a symbol to get the value, but I expect changing it now would break a lot of protocols. So, I suggest always using a symbol as a key to keep this straight when you are coding.  Incidentally, if you have a string variable `s` that you want use as a key, you can convert it by using `s.to_sym`.

2. The normal behavior is that number inputs are converted to floating point values, but the test behavior was different. 
    Along the same lines as the key type, I would suggest always giving the default value as a floating point number.
   So, `-1.0` instead of `-1`.

bjkeller, 2020-03-06:
Just to be clear this won't appear in biofab production until the system is updated.

commit d29498b7 (2020-03-06): Merge fixes into master to change table input in test to match normal behavior

fixes #390
closed 2020-03-06 by bjkeller