aquariumbio/aquarium #409: [BUG] Bad error message when protocol is missing/empty

bjkeller, 2020-06-29:
**Describe the bug**
When a protocol is missing the system gives an internal error like `server error: uninitialized constant ExecutionNamespace3JMpJ2o_Qi3Gtp5kF8GHJqqE3s8bMa1Cbhacs07dVuA::Protocol`
when it should be able to recognize that the protocol is not there.

**To Reproduce**
Steps to reproduce the behavior:
1. Create an operation type with no protocol definition.
2. Create a plan using the operation type and launch it
3. Scroll down to '....'
4. See error

**Expected behavior**
A message such as "Operation Type BLAH has no protocol defined" should be printed instead.

dvnstrcklnd, 2020-08-26:
I am unclear on what this means. I assume you would have to delete the default protocol definition?

bjkeller, 2020-08-26:
Made this more explicit. The code has to be empty – there is no `Protocol`. Though it should also happen when the class is named something else.