aquariumbio/aquarium #410: `sample.in` is broken [BUG]

dvnstrcklnd, 2020-07-08:
**Describe the bug**
`sample.in` is broken. In my hands it never returns anything. It appears to fail on the `location` condition.

dvnstrcklnd, 2020-07-08:
It's possible that this is related to running during tests, but I am not sure why that would be the case.

dvnstrcklnd, 2020-07-08:
```ruby
    inspect sample.name

    c = ObjectType.find_by(name: 'qPCR Master Mix Stock')
    inspect c

    item = sample.in('qPCR Master Mix Stock').first
    inspect item

    item = Item.where(sample: sample, object_type: c).first
    inspect item
```

```
**2020-07-08T08:45:08.107-07:00**
- {"title":"<span style=\"background-color:yellow\">INSPECTING  (String)</span>"}
- {"note":"\"TaqPath 1-Step RT-qPCR Master Mix (4x)\""}
**2020-07-08T08:45:08.112-07:00**
- {"title":"<span style=\"background-color:yellow\">INSPECTING  (ObjectType)</span>"}
- {"note":"{\"id\":23,\"name\":\"qPCR Master Mix Stock\",\"description\":\"qPCR Master Mix Stock\",\"min\":0,\"max\":1,\"handler\":\"sample_container\",\"safety\":\"No safety information\",\"cleanup\":\"No cleanup information\",\"data\":\"No data\",\"vendor\":\"No vendor information\",\"created_at\":\"2020-07-07T21:32:31.000-07:00\",\"updated_at\":\"2020-07-07T21:32:31.000-07:00\",\"unit\":\"qPCR Master Mix\",\"cost\":0.01,\"release_method\":\"return\",\"release_description\":\"\",\"sample_type_id\":11,\"image\":\"\",\"prefix\":\"M20\",\"rows\":null,\"columns\":null}"}
**2020-07-08T08:45:08.115-07:00**
- {"title":"<span style=\"background-color:yellow\">INSPECTING  (NilClass)</span>"}
- {"note":"null"}
**2020-07-08T08:45:08.119-07:00**
- {"title":"<span style=\"background-color:yellow\">INSPECTING  (Item)</span>"}
- {"note":"{\"id\":3827,\"location\":\"Unknown\",\"quantity\":1,\"object_type_id\":23,\"created_at\":\"2020-07-07T21:36:30.000-07:00\",\"updated_at\":\"2020-07-07T21:36:30.000-07:00\",\"inuse\":0,\"sample_id\":15,\"data\":null,\"locator_id\":null}"}
```

bjkeller, 2020-07-08:
It seems like this might be failing on parts of collections where the location is nil.

I would suggest using something in Item instead, since this method yields odd code.

The `where` query is close to what we would make for an alternate function, and is fine in this case.
Item does have a scope 

`Item.with_sample(sample: s)`

which allows the longer `Item.with_sample(sample: s).where(object_type: t)`

bjkeller, 2020-07-08:
fixed by eea190147053b0bccab4e21b60f86c914173c72f

closed 2020-07-08 by bjkeller