aquariumbio/aquarium #414: [FEATURE] Allow data association to external link

bjkeller, 2020-08-07:
**Please identify the user roles that your feature request relates to**

researcher, lab manager, technician, maybe protocol developer

**What is the task you want to perform?**

I want to be able to add references to externally stored data files as "uploads" without uploading 

**What is the goal you want to achieve with the feature?**

Large files, such as those from sequencing, take a lot of time to transfer. So, if a file is already stored on Illumina Basespace or a non-Aquarium managed object store.

**Additional context**

Example scenarios are any protocol involving NGS – e.g., protein design or COVID

gnomicos, 2020-08-07:
As a workaround doesn't the following work? (or am I even in the right place in Aquarium?)

Specifically, in the [ data ] tab just add a text key that is a URL.  Here is an example I added 6 keys to the StreakPlate protocol in the tutorial (see screenshot)
- "new key" is a text key that says "testing data"
- "new" is a text key that says "new notes"
- "new 2" is an upload that is an image (in this case 001.jpg - which is uploaded and copied to S3)
- "testing" is an upload that is a text file (in this case testing.txt - which is uploaded and copied to S3)
- "URL" is a text key that is a URL (in this case http://www.google.com - which is just text)

I'm sure I am missing a lot of context, such as how the info shows up downstream, etc). But in the meantime it does not appear to be a blocking issue.

p.s. On a separate note, I cannot seem to REMOVE a data item once I created it?  Hmm...

[image]

bjkeller, 2020-08-07:
We can create a data association with the URI as a string, but the goal is to do this in a way that is useful to the user so that they can get to the data without doing something convoluted.  For a file that is uploaded, there is a `download` link that allows the user to download the file.  You would not want to download one of these files b/c they are big enough you don't want to do the transfer, but the URI should be identifiable as a URI and not a string.

As far as what you are doing there, that is the operation type, which is basically the program. You would not want to attach a file to the operation type since it is specific to one execution. (Honestly, not sure why this is possible on this page). Generally a protocol will ask the technician to upload a file, and it will be associated with the item (and the plan).

bjkeller, 2020-08-09:
I remembered that the use case for this is more complex than the typical data association for an uploaded file. I also didn't describe it sufficiently. 
So, I'm going to close this and put a placeholder in the design roadmap.

closed 2020-08-09 by bjkeller