aquariumbio/aquarium #416: Several Operation Types fail after importing the workflows on the Aquarium Community page

gnomicosuw, 2020-08-16:
REPRO

1. Install a clean version of Aquarium

2. import the following community workflows (the three on the www.aquarium.bio > Community page plus the standard libraries)
- pcr-models.aq
- aq-sample-models.aq
- standard-libraries.aq
- aq-yeast-display.aq

3. Add sample Inventory (add inventory for each predefined sample type > container)
- Antibody > Antibody Stock
- Assay Buffer
- Biotinylated Binding Target > Biotinylated Binding Target Stock
- DNA Library > Labeled Yeast Library Suspension
- DNA Library > Yeast Library Liquid Culture
- DNA Library > Yeast Library in Soln 1
- DNA Library > Yeast Library Glycerol Stock
- E coli strain
- Fragment
- Media > 800 mL Liquid
- Oligo Pool
- Plasmid
- Primer
- Protease
- Yeast Strain > Yeast 50ml culture
- Yeast Strain > Labeled Yeast Strain Suspension
- Yeast Strain > Yeast 5ml culture
- Yeast Strain > Yeast Plate

4. Test each protocol using Developer > Test > [ Generate Operations ] > [ Test ] for each of the Category > Operation Types

RESULT: Some of the protocols fail. See the following list 

Next Gen Prep > Store Yeast Library Sample
    OK

Test > PCR Test
    ERROR
    could not find library 'Thermocyclers/Thermocyclers'
    Test/PCR Test: line 5: in `add'

Yeast Display > Challenge and Label
    ERROR
    undefined method `[]' for nil:NilClass
    Yeast Display/ChallengeAndLabelDebug: line 45: in `block in consolidate_protease_inputs'
    Yeast Display/ChallengeAndLabelDebug: line 45: in `each'
    Yeast Display/ChallengeAndLabelDebug: line 45: in `consolidate_protease_inputs'
    Yeast Display/ChallengeAndLabelDebug: line 26: in `override_input_operations'
    Yeast Display/Challenge and Label: line 66: in `main'

Yeast Display > Dilute Yeast Library
    OK

Yeast Display > Innoculate Yeast Library
    OK

Yeast Display > Mass Plan Data Upload 
    ERROR
    undefined method `associate' for nil:NilClass

Yeast Display > Mix Cultures 
    ERROR
    element size differs (0 should be 1)
    Yeast Display/Mix Cultures: line 50: in `transpose'
    Yeast Display/Mix Cultures: line 50: in `create_mix_table'
    Yeast Display/Mix Cultures: line 36: in `operation_mix'
    Yeast Display/Mix Cultures: line 24: in `block in main'
    Yeast Display/Mix Cultures: line 23: in `each'
    Yeast Display/Mix Cultures: line 23: in `main'

Yeast Display > Sort Yeast Display Library
    ERROR
    could not find library 'Flow Cytometry/SonySH800S'
    Yeast Display/Sort Yeast Display Library: line 7: in `add'

dvnstrcklnd, 2020-08-25:
`Test > PCR Test` requires https://github.com/aquariumbio/thermocyclers.

`Yeast Display > Challenge and Label` has some expectations about attributes of samples and items that must be met. This should fail more gracefully.

`Yeast Display > Mass Plan Data Upload` uses file uploads which generally don't work in tests run through the GUI.

`Yeast Display > Mix Cultures` requires specific parameters. This should fail more gracefully.

`Yeast Display > Sort Yeast Display Library` requires https://github.com/klavinslab/flow-cytometry, which should be move to the aquariumbio repo

dvnstrcklnd, 2020-08-26:
I'll work on bringing this documentation up to date.