aquariumbio/aquarium #418: Aquarium displays entire history of key-value pairs for data associations

dvnstrcklnd, 2020-08-20:
**Describe the bug**
When a data association with a given key is updated, the system evidently creates a new row for that key in the database rather than updating the existing row. This is good from a record-keeping perspective, but the system does not remove duplicates when displaying the associations through the GUI. This results in confusion about which value is the correct one for a given key. The problem is particularly salient when iteratively adding provenance to an item, and results in lists of associations like this one:

[image]

**To Reproduce**
Add multiple data associations with the same key to any object.

**Expected behavior**
Following usual conventions for key-value data structures, retrieving data associations for an object should yield no more than one data association for a given key. It should be the one most recently *updated* in the database.

Additionally, when adding data associations through the GUI, it would be good if Aquarium provided a warning when a key is being duplicated / overwritten.

dvnstrcklnd, 2020-08-20:
@bjkeller @gnomicosuw @marikoja It would be good to fix this with high priority. It is a really big problem when using provenance associations in the context of the COVID protocols.

bjkeller, 2020-08-20:
I think this is part of a broader problem of being unable to represent the state of an item. You should be able to write a protocol that simply modifies an item without having to worry about provenance being right. 
For provenance, we need to distinguish the state of a well before and after it has been modified, but for the protocol the author should only care about the current state. 

My question is whether we should go ahead and address this root issue, or do the filter?

dvnstrcklnd, 2020-08-20:
This issue is intended to be independent of what is happening in protocols,
regarding provenance or otherwise. It's confusing to have duplicate
associations for a given key, regardless of how they are generated.

As I indicated, addressing this issue would mask the separate issue that is
happening in the protocol code, but addressing that issue would not fix
this one.

dvnstrcklnd, 2020-08-20:
Regarding timing, as I indicated I think this is an urgent issue. I also
think it can be addressed with relatively small changes, and the result
would be a more faithful representation of the "current state" of the item.
While I like the idea of addressing the root cause, I don't see that
happening on an acceptable timeline,

bjkeller, 2020-08-20:
OK, sounds good. 

If I'm understanding correctly, we have two stories here:
1. As a user viewing the data associations for an item, I expect to see the value most recently assigned to a key in order to understand the current state of the item.
2. As a user creating a data association, I want to confirm when the key exists in order to avoid mistakenly changing the displayed state of the item.

Is that correct?

dvnstrcklnd, 2020-08-20:
Yes

marikoja, 2020-08-20:
That all makes sense to me, this ticket will address the first story that Ben restated.

bjkeller, 2020-08-20:
@marikoja Can you make a new issue for story 2 and cross link with this one? 

Also, would be great if you can make sure both are exposed as TODOs on https://github.com/orgs/aquariumbio/projects/1

bjkeller, 2020-08-20:
Associations are listed in multiple ERB files using calls to `DataAssociator::data_associations`:
- app/views/plans/_sidebar.html.erb:    `<div ng-if="plan.data_associations">`
- app/views/data_associations/_template.html.erb:         `ng-repeat="da in record.data_associations track by $index"`
- app/views/data_associations/_list.html.erb:      `<li ng-repeat="da in parent.data_associations | filter: { key: '!notes' }">`
- app/views/operations/_operation_list.html.erb:            `<li ng-repeat="da in op.data_associations">`
- app/views/operations/_operation_list_short.html.erb:            `<li ng-repeat="da in op.data_associations track by $index">`
- app/views/items/show.html.erb:                 `ng-repeat="da in part.data_associations"`

bjkeller, 2020-08-20:
I think the solution is to add a `current_value` scope function to `DataAssociation` and use the scope in the queries in `DataAssociator::data_associations`

@gnomicosuw do you have time for this before you go?

dvnstrcklnd, 2020-08-20:
Note that most data association methods in the API return single keys, but
this may be an accident of building the hash.

gnomicosuw, 2020-08-20:
@bjkeller - looking at this now and will check something in for us to review shortly. I'm not sure I have a great repro scenario so maybe we can have a quick call to review after I update it

gnomicosuw, 2020-08-20:
As Devin suggested, every time a key value gets changed it appears a new row is written to the DB. 

To return the latest value for each key it appears we can get the value for each key based on the latest 'id' field fr that key (as opposed to the latest 'updated_at"'field). From a query perspective that is faster and more robust (among other things it is possible, although unlikely, that two records in the DB have the same 'updated_at' value)

As a brute-force alternative we can also simply query for all values and loop through them to get the latest one for each key.  This might be the best solution since I assume there are not that many key-value pairs for each sample.

@bjkeller - do you have a preference?

bjkeller, 2020-08-20:
Yes, I should look more.  

There is another method `DataAssociator::associations` that builds a hash from the `data_associations`.
It is used in `app/views/operation_types/_test.html.erb`. 
Maybe can just call that.

gnomicosuw, 2020-08-20:
Yes. That is the method I am looking at!  Wanna zoom quickly and I can show you my ideas?

bjkeller, 2020-08-20:
There are direct `DataAssociation` queries in

> As Devin suggested, every time a key value gets changed it appears a new row is written to the DB.
> 
> To return the latest value for each key it appears we can get the value for each key based on the latest 'id' field for that key (as opposed to the latest 'updated_at' field). From a query perspective that is faster and more robust (among other things it is possible, although unlikely, that two records in the DB have the same 'updated_at' value)
> 
> As a brute-force alternative we can also simply query for all values and loop through them to get the latest one for each key. This might be the best solution since I assume there are not that many key-value pairs for each sample.
> 
> @bjkeller - do you have a preference?

preference: faster and more robust

dvnstrcklnd, 2020-08-20:
I think grabbing the latest ID is OK only if you can be sure that rows are
never updated.

gnomicosuw, 2020-08-20:
I cannot be sure that rows are never updated. In fact I did a quick check in the DB and the timestamps are not always the same. It could be the key is created and then saved (vs updated later) but I do not have enough knowledge of the rest of the system to guarantee that.

For now I suggest just looping through the values.  I can always take another look at it later.

bjkeller, 2020-08-20:
There are `DataAssociation.where` queries in 
- CollectionsController
- OperationsController
- UploadsController
- Collection

bjkeller, 2020-08-20:
`CollectionsController::save_data_associations` updates a DataAssociation found by ID, I think this is only called from the collection editor view

dvnstrcklnd, 2020-08-25:
If this fix covers most ways of accessing the associations, including through Trident, then yes I believe this is what I want.

bjkeller, 2020-08-25:
I believe trident is using the JsonController to grab `DataAssociator.associations`.  Will verify.

commit 4373bfb1 (2020-09-03): Squashed commit of the following:

  * Make associations methods use recent filter.
    1. Move filter method from json query result class to data association class.
    2. Change data associator associations method use filtering method.
    3. Change names of scope and filter methods.

  * Remove limit on query
  * Change data association queries in JSON Controller to method call
  * Replace data association query
  * Move data association details into model class

  * Add duplicate key filter to JSON query
    - checks for data association query
    - filters duplicate keys selecting most re [... cut]
commit 67faf748 (2020-09-03): Query current association (#420)

* Add most_recent scop method to DataAssociation

Add most_recent scope method to Data Association:
- orders first by descending update time
- then by descending record id

* Add spec to test filtering duplicate keys in DataAssociations

* Add duplicate key filter to JSON query

Add duplicate key filter to JSON query
- checks for data association query
- filters duplicate keys selecting most recently updated association

Fixes #418

* Move data association details into model class

* Replace data association query

* Change data association queries to method c [... cut]
closed 2020-09-03 by bjkeller