aquariumbio/aquarium #419: cannot pass RAILS_ENV config setting to docker-compose

gnomicosuw, 2020-08-20:
For example, I tried running rake db:migrate to run the database migrations on my dev box

REPRO
$ docker-compose exec app RAILS_ENV=development rake db:migrate 

RESULT
ERROR: OCI runtime exec failed: exec failed: container_linux.go:349: starting container process caused "exec: \"RAILS_ENV=development\": executable file not found in $PATH": unknown

NOTE: docker-compose is already set up to run in my development environment, so the following worked just fine.
$ docker-compose exec app rake db:migrate

bjkeller, 2020-08-20:
This command uses the `entrypoint.sh` script with a shell command `exec "$@"` (different with `docker-compose exec`).

it seems you have to precede each environment variable with `env`

so command would be
```bash
docker-compose exec app env RAILS_ENV=development rake db:migrate
```

bjkeller, 2020-08-20:
I changed the instruction in https://github.com/aquariumbio/aquarium/commit/d01ab3b4758d201ca4be0c2125cb873b3d8e3e3d

But need to confirm that this works before closing this