aquariumbio/aquarium #441: [BUG] Media output doesn't display container type in pending and scheduled states

spcoon, 2020-10-29:
**Describe the bug**
For media operations, the display in the operation table (pending and scheduled) has incomplete out: data. Currently, it presents a default 'media' as container type, rather than the actual container type selected in the operation.

**To Reproduce**
Steps to reproduce the behavior:
1. In Nursery, as yourself, Go to 'Aquarium > Designer Tab > Design > Operation Types > Reagents > Make Liquid/Agar Media'
2. Click on 'The output circle on the operation box. In the media output field below, type in SDO and then choose any result. The left pane will default to a container type (this is the data that we want to display in the out: column in the operation tables). Click 'save' then 'launch' then check the budget checkbox and submit.
3. Go to 'Manager tab' and click on 'Reagents' in the left nav. Click on the pending cell for 'Make Liquid/Agar Media'
4. See error: The 'out:' data for container type only displays the term 'media'

**Expected behavior**
The 'out:' data for container type should display the container type chosen when setting up the operation in the Designer tab. In my case, it was, "800 mL Agar." Without this discernment, multiple media operations submitted on the same day across the same user are hard to discern and causes the manager to jump into plans to figure out which is which.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Computer/Device (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

bjkeller, 2020-10-29:
"Media" is the name of the output not the container type

spcoon, 2020-10-29:
"media" is too general to distinguish from other operations.

bjkeller, 2020-10-29:
Sure, but this seems like a design issue rather than a bug. Aquarium doesn't present the object type of either inputs or outputs.

My general view is that media is a special case, and should be handled differently than by forcing it  into the mold of other protocols.
However, could also see two possible user stories that could address this

> As a lab manager, I want to be able to view object types for media requests so that I can create jobs by media sample and the container type. 

or

> As a lab manager, I want to be able to group operations by input/output types, plan, user (project, ???)  in order to create coherent jobs quickly

spcoon, 2020-10-29:
right, it isn't a bug. thank you. i'm glad we've navigated that flub.

how difficult would it be to add the container type string? (e.g. Media : **800 mL Agar** : SDO -Ura + G418) that's what aza said would be helpful.

bjkeller, 2020-10-29:
It looks like this page already has the code that will display the object type of the output item, but the output item does not yet exist when the operation is pending.
I think we could just add a conditional that checks that the operation is pending and then uses the properties of the output (which is distinct from the item)

So, in principle, not hard

spcoon, 2020-10-29:
sounds like a plan to me.