aquariumbio/aquarium #469: Cannot preview completed jobs [FEATURE]

malloc3, 2020-11-23:
**Please identify the user roles that your feature request relates to?**
Lab Managers, Protocol Development

**What is the task you want to perform?**
In previous version of Aquarium manager page, Under 'Error' and 'Done' you could navigate to the job and see exactly how it was performed.    The functionality no longer exists (or I could not find it).     This functionality is incredibly helpful for debugging protocols and for reviewing jobs etc.

**What is the goal you want to achieve with the feature?**
I need to be able to preview jobs that have been completed (either "Error" or "Done") and walk through the step by step views that the technician saw.

bjkeller, 2020-11-23:
The job header is not included for done and error and so the job link is missing

bjkeller, 2020-11-23:
Also affected by #458

marikoja, 2020-12-02:
@malloc3 Confirming the ask is to put the jobs column back into the table only when  **error**ed or **done** operations?
[image]
[image]

malloc3, 2020-12-02:
Yes.   Or equivalent functionality.

PR #479 (2020-12-04): Tech Handoff Bugs
Several small changes to address various bugs/requests
Fixes #458 
Fixes #469 
Fixes #471 
Fixes #472
commit aeeac92f (2020-12-04): Tech Handoff Bugs (#479)

Address issues from tech-handoff changes:
* Update styling of the job number link (Fixes  #458).
* Allow all users to mark items as deleted (Fixes #471)
* Show Jobs column for completed jobs (Fixes #469)
* Sort jobs by job id (#472)
closed 2020-12-04 by bjkeller