aquariumbio/aquarium #487: [FEATURE] Compact I/O list for operations in manager

dvnstrcklnd, 2020-12-09:
When planning operations for the Duke COVID testing workflow, we need to be able to plan operations with 96 input samples each, and batch 4 of these operations together based on the plan or output sample. The list of 96 input samples is mostly uninformative, and necessitates a lot of scrolling to find the 4 operations to be batched:

[image]

I would like to be able to select all 4 operations without scrolling. This will streamline the batching of operations and make the process less error-prone. 

This is a fairly high priority because we anticipate running this workflow at scale early in the new year.

spcoon, 2020-12-09:
for this exact use case—where you want to select all four operations without scrolling—the "all" button at the top of the interface selects all four operations. clicking "schedule" batches them into a job.

with technician handoff, the all button is replaces by a checkbox that selects all operations.

dvnstrcklnd, 2020-12-09:
Clarification: I mean all 4 operations _that I intend to batch_, not all operations in the list.

spcoon, 2020-12-10:
ok, so not this particular case since there are only four operations.

i need you to describe what's necessary operation information and what's not. there's not much design thinking to do here aside from simplifying the display. you said "mostly uninformative" above. what parts are informative?

bjkeller, 2020-12-10:
It seems like it might be a good idea to hide a long I/O list by default.

spcoon, 2020-12-10:
i agree, ben. what form do i need to fill out to specify a solution of "if more than five lines, display **more** link that expands on click"

?

dvnstrcklnd, 2020-12-11:
The plan and the output sample are what's informative. The intent is that they will be batched on the output sample name, which I intend to have a 1:1 relationship with the plan.

I am agnostic as to whether we implement a filter function or collapse the list by default. The latter seems like it would be easier, but probably not as generally usable and thus less desirable in the long-term.

spcoon, 2020-12-11:
ok, i re-read the original post. this is a bit trickier than what i originally thought. i'm going to prioritize this for monday.

spcoon, 2020-12-15:
@dvnstrcklnd i suggest we just add sorting to this interface and truncate operations that generate > five lines. not sure what other quick fixes would make sense.

bjkeller, 2021-01-06:
This is v2-tech-handoff, correct?

marikoja, 2021-01-19:
@dvnstrcklnd @bjkeller @spcoon this is the simplest solution I could come up with that I think solves for Duke's need. 

Inputs start hidden:
[image]

Toggle switch shows inputs: 
[image]

dvnstrcklnd, 2021-01-20:
I think this checks the box for the Duke scenario, although it might not be the most future-proof way to do it.

marikoja, 2021-01-20:
@dvnstrcklnd I agree with you about that this solution may not be future proof. However this code is in angular and will be re-written into react I feel confident this is a solution that minimizes additional tech-debt.

PR #530 (2021-01-29): Bugs tech handoff
* Fixes #489: Tech assignment menu size handles single technician in list
* Fixes #462: Title size
* Resolves #487: Compact IO list