aquariumbio/aquarium #506: [BUG] Production: the { hamburger > location wizards } page does not scroll

gnomicosuw, 2020-12-21:
repro:

1. log into aquarium as user with with admin privileges
2. navigate to { hamburger > location wizards } page
result: the page does not scroll (makes it kind of challenging to edit a location wizard that is below the fold)

(see screenshot - notice there is no scrollbar)

[image]

note: you can add overflow-y:scroll to the <div layout="row" class="layout-row" style="overflow-y:scroll"> div in the view and get the content to scroll.  It might be nicer to scroll the table contents and keep the title, but that also affects the vertical space.  And since this is all probably going to change later I would do whatever is easiest. (It has been this way for a long time and as far as I know no one has complained).

Here's another screenshot with the overflow setting (notice the scrollbar)

[image]

bjkeller, 2020-12-22:
Is this problem present in either the master or staging branch?

gnomicosuw, 2020-12-22:
It is in both.

bjkeller, 2020-12-22:
OK, let's update this in tech-handoff

gnomicosuw, 2020-12-22:
There is a whole bunch of Angular and/or React stuff on this page which I did not want to try to decipher, so I just hacked in a solution.
- added a variable { @scroll = true }
- added inline styling { style="overflow-y:scroll" } if @scroll is true

Note that the "Budgets" page does scroll, so there is probably some setting buried somewhere.  But the hack works.

I checked it in under the branch "scroll" off of "master". I will submit a PR.

bjkeller, 2021-01-06:
fixed by #507

closed 2021-01-06 by bjkeller