aquariumbio/aquarium #561: Sort/set order of OperationTypes in Manager Tab[FEATURE]

malloc3, 2021-02-19:
It would be nice to be able to either sort "alphabetically", "most common use", or specify a specific custom order.

The current order feels arbitrary and confuses people.   If we could order based on which Operation Types are executed first that would be really neat.

[image]

malloc3, 2021-02-19:
@spcoon  Ben told me to tag you here.

bjkeller, 2021-02-19:
Not that this is good practice, but illustrating people's workarounds:
there are some historic operation types with names starting with numbers
representing the order in the canonical workflow.

malloc3, 2021-02-19:
Oh smart.  In this specific case that may be a great idea.

spcoon, 2021-02-22:
@malloc3 - i understand the notion of "Operation Types [that] are executed first," but i do have questions.

would this specific list of operations/protocols ever change? not just in this particular grouping, "RT-qPCR," but in other groupings as well? 

could other workflows be coded with the same grouping, thereby making chaos out of order? (i'm assuming the grouping assignment happens in the protocol code). 

might other labs use this grouping taxonomy approach in the same manner as the biofab? note: we might want to reconsider the grouping notion altogether once we get the new design system into play.

if there's more flexibility involved, does the hierarchy of operation status (pending, waiting, etc.) give affordance as to what should be worked on first?

malloc3, 2021-02-22:
Well that is the hard part I suppose.

The specific list of operations/protocols would most certainly change.   To be clear this ordering of these operation types on this page would not affect the protocol code or what the technician sees while running things.

The desired order could completely depend on the lab, workflow, or any number of arbitrary conditions.

Perhaps in stead of sorting based on operation types the list could be sorted based on operation status.  E.g. Operation Types with a "Pending" status are at the top etc...

spcoon, 2021-02-23:
@malloc3 

here's how we're changing the interface, making pending highest priority, then errors (to make pending once again), then waiting, etc.

[image]

malloc3, 2021-02-23:
That looks perfect!   Solves the exact issue.

spcoon, 2021-02-23:
@malloc3 i'm going to screenshot this moment of perfection feedback and savor it. :)