aquariumbio/aquarium #562: Can not specify a collection as an input [BUG]

malloc3, 2021-02-19:
Researcher, Manager, Developer

Specify a WHOLE collection as an input/output to an OperationType.  It is currently only possible to specify an Item or a Part (of a collection).   This causes confusion on multiple levels when workflows operate on a whole plate.

Being able to specify a whole collection would eliminate the use of hacks that can be confusing to users in the Researcher and Manager roles.

bjkeller, 2021-02-19:
Is the collection not an item?

malloc3, 2021-02-19:
I think it's an extension of the `Item` class yes.    If you have the **OperationType** which accepts an Item I do not think you can pass it a `Collection` (if I am wrong on this then YAY).

You can have the **OperationType** specify a `Part`  but that is a specific `Part` within the `collection`.

Typically to get around this I have it specify a `Part` and then within the protocol I have `op.input(INPUT_PART).collection` to get the whole collection.   This is a non issue from a development stand point.  

However when in **Designer** you have to specify a specific `Part` within the collection and "assume" it will take the whole collection.   This can be pretty confusing to anyone who isn't super familiar with that process.   Additionally I feel like "assuming" that the protocol will just take the whole collection is bad practice.

bjkeller, 2021-02-19:
Can you look at the YG_Harmonization operation types on BIOFAB/production
and explain how that is different from what you are describing here?

bjkeller, 2021-02-19:
maybe the problem is that collections don't have sample types but the
system is expecting one

malloc3, 2021-02-19:
Maybe that is the root of the issue indeed.