aquariumbio/aquarium #620: aquarium-local hangs when batching operations

dvnstrcklnd, 2021-04-21:
My aquarium-local instance hangs at `Loading operations ...` when I batch operations.

dvnstrcklnd, 2021-05-24:
I can see that the status of the operations are changed to `scheduled` but the table of batched operations does not load in the manager tab.

bjkeller, 2021-05-24:
Is this with a copy of the production database or with a database that you built from scratch?

dvnstrcklnd, 2021-05-24:
Define "from scratch."

dvnstrcklnd, 2021-05-24:
I'm adding stuff to the default database.

dvnstrcklnd, 2021-05-24:
OK, I think this is happening:
```
app_1           | ActiveRecord::StatementInvalid (Mysql2::Error: Table 'production.view_job_assignments' doesn't exist: SHOW FULL FIELDS FROM `view_job_assignments`):
app_1           |   app/controllers/operations_controller.rb:148:in `block in manager_list'
app_1           |   app/controllers/operations_controller.rb:147:in `each'
app_1           |   app/controllers/operations_controller.rb:147:in `manager_list'
```

bjkeller, 2021-05-24:
'from scratch' => from default database

bjkeller, 2021-05-24:
need to look at default database and make sure has views

dvnstrcklnd, 2021-05-24:
It doesn't.