aquariumbio/aquarium #657: Display of special characters

dvnstrcklnd, 2021-05-26:
**Describe the bug**
Special characters encoded with HTML don't display properly in some parts of the interface.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a location to an item using a special character, e.g., "37 \&deg;C shaker"
2. View from show blocks -> OK
3. View location in technician view or item view -> bad

**Expected behavior**
Should display uniformly.

**Screenshots**
[image]

[image]

[image]

bjkeller, 2021-05-26:
Can you share the show block?

dvnstrcklnd, 2021-05-26:
```
    show do
      title 'Resuspend and Store Samples'

      note "Resuspend each sample with 30 #{MICROLITERS} #{FORMIC_ACID_WATER}."
      note "Store the samples at 4 #{DEGREES_C}."
    end
```

dvnstrcklnd, 2021-05-26:
Where DEGREES_C = &deg;C

bjkeller, 2021-05-26:
Relevant code:

show block "note":
https://github.com/aquariumbio/aquarium/blob/c9dcfd23dc6e5b3e4b7bedf7b8000a30dbb42c68/app/views/technician/_show_block.html.erb#L6

show block "take": https://github.com/aquariumbio/aquarium/blob/c9dcfd23dc6e5b3e4b7bedf7b8000a30dbb42c68/app/views/technician/_show_block.html.erb#L144

item popup location field:
https://github.com/aquariumbio/aquarium/blob/c9dcfd23dc6e5b3e4b7bedf7b8000a30dbb42c68/app/views/items/_template.html.erb#L117

bjkeller, 2021-05-26:
The location in the item popup is a textfield – meaning you can edit it. The others are just display.

This gem might help: https://github.com/threedaymonk/htmlentities